joshr120/PD-Stepper #3: GREAT IDEA! But more to add to it!!!!!

Cyrille-de-Brebisson, 2024-08-27:
Hello,

I LOVE YOUR IDEA. I just ordered one of the "standard" closed loop controler for a nema17, but yours is definitely one step above...

However, I have some requests/additional ideas here...

I am an amateur astronomer, and I make telescopes.
Telescope typically need 3 to 4 motors to work. 2 for moving the telescope, 1 for moving the focus and one for the filter wheel.
One issue that we have ALL THE TIME is cable managements. A typical astrophotography setup is a cable mess!

So, here is the request, Could the Aux connector be reworked so that it has:
- Data AND power
- They are 2 of them, one in, one out, in order to allow to daisy chain motors?

Thanks,
Cyrille

joshr120, 2024-08-27:
Hi Cyrille,

Thanks for the ideas, those particular connectors being used for AUX are only rated to 1 amp so would not have enough power to daisy chain multiple motors together. If you had multiple motors all running at full power powered off one USB PD port you could reach the max power draw pretty fast. 
However, if you really wanted to daisy chain them there are solder pads on the underside where you can access/inject the PD voltage directly (just make sure you don't use the USB connector at the same time). Using I2C you could then daisy chain the data with the existing AUX and qwiic connectors . You could also use something like ESPNOW if you wanted to do this wirelessly.

Cyrille-de-Brebisson, 2024-08-27:
Hello,

I understand...

- Is it possible to have nice easy to solder vias for power daisy chain, with 2 of each (2 gnd, 2 power)? to make it easy and stable? something like a 2*2 1/10" header type thing?

- I see that you have a number of LED on the system. Some seems to be programmable (great), but others seems to be always on type thing. It is possible to have them controlable so that they can be turned off? As you can imagine, LEDs are not what you want near a telescope :-)

Cyrille

joshr120, 2024-09-18:
There is not currently room for extra power headers on the board at the moment, the solder pads on the underside are the best I could fit to stay within the NEMA17 form factor.

The additional LEDs are not currently set up to be able to be turned off in this version, although they are derated to not be full brightness. They can also be removed (or covered) if they are not needed.

fladdie, 2024-09-26:
Can bus would be a great feature

001oc, 2024-10-10:
Kicad: https://github.com/user-attachments/files/17332545/PD-Stepper.zip

EasyEDA: https://oshwlab.com/adamscamden2/pd-stepper

dwmw2, 2025-02-10:
What about expanding the motor connector to 6 pins to allow for power and ground through that?  I've just been playing with feeding power from behind, using S1761-R42 terminals. It works, but it's a real pain to solder up and assemble by hand, even in a mount 3D-printed for the purpose.

[image]

Using Veroboard is just as painful to solder but at least gives a better mechanical connection.

[image]

I'm tempted to get some tiny board professionally manufactured. Wonder if they'd be useful to include as an option in the kits? It could even be done as a full sized 42.3x42.3 board with the spring contacts on top and the PH connector on the bottom, designed to sit under the metal spacer/heatsink if the full kit?

joshr120, 2025-02-10:
Expanding the motor connector to 6 pins would mean that the motor loom could be plugged in the wrong way around (as the Nema 17 also has a 6 pin PH). This would also require the molds to be modified for the injection molded covers.

I do quite like the pogo pin idea.  The only problem I see with having the connector underneath like you do is it affecting the spacing between the magnet and encoder.  I guess you could add a spacer underneath the magnet or stack another magnet on top. 
However a custom board could possibly replace the aluminum spacer and bring the connector out to the side and would not affect the form factor too much as you would already have wires coming out.

[image]

This mock up has it in front of the USB connector which would be a pain for programing but you really don't want both plugged in at the same time and this could fry your computer. This uses a 1mm board + a 1.6mm board to get the spacing as well as allowing a recess for the pogo's to sit in. Let me know what you think.

Could also be nice to use something which can carry a bit more current than a PH connector so that daisy chaining power could be an option.

dwmw2, 2025-02-10:
If I wanted it that side I'd just use the USB connector. I'm trying to keep the wires all on one side. This is the prototype version (inward facing, as I'm not allowed to dig holes in the wall until the proof of concept is working).

[image]

I'm trying to avoid that extra cable at the top. (Those green LEDs are too bright too but I'll try to hide those with a black cover).

I have my magnet mounted inside the gear, protruding through the piece of the mount that the power connectors are in, so that part works OK here. And I've found that soldering the spring connectors to a Dupont header designed for board-mounting seems to work...

[image]

joshr120, 2025-02-11:
Ah I see, did not realize your goal was to keep everything on one side. I thought you just wanted a way to power it with straight DC rather than USB PD.  Either way a custom PCB or 2 would probably be the way to go.

dwmw2, 2025-02-11:
The spacers require about 3mm between their base and the underside of the PD stepper board, so I think I'm happiest with my setup of soldering them to a Dupont header inset into the mount, rather than the thickness of an extra PCB.

But since someone else had already asked, I figured it was worth mentioning the possibility of a second PCB to supply power.

FWIW I also just realised I *do* need a plug at the top for the AUX pin which will be connected to the reed switch at the top of the blind, for homing. I'll probably bring that single wire straight down into the same plastic piece and bring it to a 3-pin header with the Gnd and Vbus I already have there. Are there unused GPIO pins on the ESP32? Maybe the QWIIC connector could have its *own* pins for its own i2c bus (or other purposes). The i2c bus with the as5600 encoder is going to be permanently busy with polling anyway, isn't it? Not much bandwidth for anything else.

joshr120, 2025-02-11:
Currently the QWIIC connector is on the same I2C bus as the as5600 as I was getting low on pins and that is how most of the QWIIC ecosystem works. If the encoder is being polled very fast it may use a fair bit of bandwidth but would also use a far bit of CPU bandwidth at the same time while reading, so a second bus may not be of too much use other than if there are address conflicts or you are getting fancy with multitasking. 

I may look into putting it onto its own bus in a future version, but for something like a limit switch this is exactly why the AUX connector is there.

dwmw2, 2025-02-11:
The encoder needs to be polled fast, or it misses revolutions. I've been playing with it, and ESPHome going out to lunch seems to be the reason why my tracking was wrong. I occasionally see warnings like

```
[09:19:26][W][component:237]: Component as5600.sensor took a long time for an operation (4184 ms).
[09:19:26][W][component:238]: Components should block for at most 30 ms.
```
So the blind closes correctly because it did calculate the correct number of steps to turn the motor, but the encoder tracking only thinks it's closed part way. So when it's told to open again, it only goes part way.

But then #22 happened while I was experimenting and now I don't have a laptop so it'll be a little while before I can experiment some more...

joshr120, 2025-02-11:
yes you need to be reading fast enough to catch a wrap around. Just did some quick timing with Arduino and a read takes ~500 microseconds so you would need to be polling around 2000Hz to max out the bus.  The max speed of a stepper is _relatively_ slow so you should never run into the problem of not being able to poll fast enough to catch a wrap around. 
It does sound like your problem is ESPHome related. It may actually be polling too fast and hogging the CPU which ESPHome may not like, you could try slowing down the polling rate.

dwmw2, 2025-02-11:
I think I'd already backed the poll rate down to 50ms from the originally configured zero, after seeing ESPHome watchdog resets because the idle task was never getting to run. That should have been fine.

closed 2025-05-28 by joshr120