joshr120/PD-Stepper #4: KICAD fils available

001oc, 2024-08-30:
great idea you realized, matching a project I was looking a solution for some time now. I would like to expand the projekt abit for my needs are KICAD files (or something else) available?

Thanks

Johnr24, 2024-09-10:
Yes, +1 Please supply the kicad or original source files. You can't truly call your project open-source if you aren't providing the easiest and least resistive way of allowing users to make changes and alterations...

joshr120, 2024-09-18:
The PCB Source files are not available at this stage sorry, feel free to use the schematic and GERBERs to base your project on.

closed 2024-09-18 by joshr120

Johnr24, 2024-09-18:
this is not compliant with GPL see section 7. Gerbers and Schematics are effectively compiled binaries and not source files,

001oc, 2024-09-18:
I think it would and should be his decision the way he feels it would be open source or something similar ... and we have to accept it instead of poking arround. I think it is well made and I dont blame someone to try to make some money to get back its efforts making some valuable pcb. And its way not to complicate to rebuild it - what I am doing right now. I am doing currently a rough "rebuild" of the pcb. Its purpose should be to help that people which want do expand or modify the good base idea - as me. I think that this is not disturbing the business of @Johnr24 as the onces that want to modify the pcb even are no potential customer of @Johnr24 - I think that could be a point of view even for @Johnr24 - but as I sayed, not my decision.

I am happy for the material that is already posted and it helps me to redesign the pcb quit quickly I think - from my point of view this is more than enough and I am thankful for that.

SoupySoups, 2024-10-10:
I know im gonna get some backlash for easy eda but i have recreated the board with my own differences [here ](https://oshwlab.com/adamscamden2/pd-stepper). Everything works great from my tests.

001oc, 2024-10-10:
Thanks alot to @SoupySoups .

I have imported your stuff to Kicad roughly.

From my quick overview it seems to be working - at least it is a proper base to work further. 

[PD-Stepper.zip](https://github.com/user-attachments/files/17332545/PD-Stepper.zip)

Johnr24, 2024-10-10:
Excellent. Amazing work!  Could I suggest starting a fork of this project
that is truly and correctly open source?

tmarkson, 2024-10-10:
The fact that you could redraw Josh's design from the files that he provided suggests the project is open enough already!

joshr120, 2024-10-11:
Now that pre-orders are starting to be shipped I have published the PCB Files (Altium format). 
I also had a quick look at @SoupySoups easyeda project and can offer a few pointers which will help it work better if someone plans on using those files.

Main reason I did not release them earlier was to stop someone simply slapping their logo on it and beating me to market while I was ramping up production & waiting for molds to be made. (Full kits will now come with injection molded polycarbonate covers and a black anodized heat spreader/spacer).

SoupySoups, 2024-10-11:
> Now that pre-orders are starting to be shipped I have published the PCB Files (Altium format). I also had a quick look at @SoupySoups easyeda project and can offer a few pointers which will help it [...]

I understand your reasoning, I recreated the design not to sell the product but to put in a project i am currently designing in which i needed multiple of these steppers. I understand the extra value that you provide other than the raw pcb but i did not need the extra parts and the cost for 8x kits per unit of my design just did not make economic sense for my application. 

The copy in easyEDA was rushed and im not that great at PCB design but all functionalities (i removed some of your features that I didn't need to further save costs and make layout easier, aka. buttons and expansion). Everything seems to work well though.

In all, yeah, your board is reeeeeally nice. I wish i had your skills, and I agree, mine is not meant to be sold in a kit, its rough.

> The fact that you could redraw Josh's design from the files that he provided suggests the project is open enough already!

In response to this, its tough, wouldn't call having to go through grbl files manually "open".

Josh I appreciate this and am thankful for what was shared, even if difficult. I notice you dont have any example code with measuring vbus voltage, using the temperature sensor, or using the AS5600 in combination with the TMC2209 for closed loop stepper control. If you would like, I would be more than happy to share my code, implementing such features.