joshr120/PD-Stepper #6: Kicad and EasyEDA files available now...

001oc, 2024-10-10:
For further development there are files for Kicad and EasyEDA available:

Kicad: https://github.com/user-attachments/files/17332545/PD-Stepper.zip

EasyEDA: https://oshwlab.com/adamscamden2/pd-stepper

Cyrille-de-Brebisson, 2024-10-11:
Hello,

One request, if possible. Would it be possible to add 2, 2 pins headers with GND and vbus?

The aim would be to be able to bring power to stepper "one" (either using USB or a 12V power supply), and then daisy chain 3 or 4 steppers from there. I do not have a huge power consumption (around 1A for the 2 main motors, who are only running for short intervals), and around 0.5A for the 2 others which will run around 1 minute per night each...

I tried looking at the PCB to see if I could do it myself, but I am way not good enough to start playing with such a non-trivial PCB (is it a 4 layers PCB?)

Thanks,
Cyrille

001oc, 2024-10-11:
> Would it be possible to add 2, 2 pins headers with GND and vbus?
I think there seams to be enough free space to do so. Anyway, I wouldn't do that this way,

>  (is it a 4 layers PCB?)
no, its a "simple" 2-layer pcb

Cyrille-de-Brebisson, 2024-10-11:
Hello,

>>   Would it be possible to add 2, 2 pins headers with GND and vbus?
>I think there seams to be enough free space to do so. Anyway, I wouldn't do that this way,
How would you do it?

I am trying to use these motors to make a telescope mount...
Telescopes have WAY to many cables! this should give you an idea of the mess that you can get into :-) https://i.ytimg.com/vi/Ip0Ok9OncNY/maxresdefault.jpg

Anyhow, on the other hand, we DO have 12V available to us as all astro equipement uses "raw" 12V...
Typically, we use a slow discharge car battery with a 12V boost converter to generate said 12V...

However, finding a USB-C power supply (portable) that will have 4 PD output and will run on 12V would be quite hard...
Plus it will cause tnagled wires as the scope moves..

Hence the idea of being able to connect the motor in a daisy chain (or distribution tree as needed) in order to reduce cables and cable snags.

Cyrille

joshr120, 2024-10-11:
Now that pre-orders are starting to be shipped The Altium Source Files been added to the repo.

closed 2024-10-11 by joshr120