joshr120/PD-Stepper #8: Not a problem, but a question

Cyrille-de-Brebisson, 2024-10-14:
Hello,

I have noticed with the traditional drivers that microsteps are NOT all equal.

For example at 1/8th using an A4988, the various microsteps are definitely not 0.225° appart (360/200*8). Although 8 microsteps do indeed equal 1.8°
DRV8825 at 1/32 are much better, but even then, microsteps are not concistent...
(These are tests that I did using a mirror on a motor and measuring the displacement of a laser 5m away)...

So, here is the question. In your design you have a TMC2209 with 256th microsteps, but also the 14 bit position sensor (which give the equivalent of a 1/80th microstep precision, and is, itself not truly linear according to the datasheet)...
Would it be possible to generate a 51200 entry table that gives, for each of 200*256 microstep, the sensor reading in order to characterize the real position for each microstep?

Thanks,
Cyrille

joshr120, 2024-10-15:
Hi Cyrille,

The "microstepping consistency" of the TMC2209 should be a lot more accurate than the A4988. 

To answer your question I imagine the characterization you are looking for would likely change depending on a number of factors such as static friction of the motor, winding quality, how centered the magnet is on the encoder, etc. The linearity of the AS5600 will also depend quite a lot on how centered the magnet is on the sensor (Although this would stay consistent for each setup so could be calibrated out) 

However, I have done a quick test as you asked and made a table showing encoder reading vs microsteps. (400*256 = 102400 data points as my motor was a 400 step per revolution stepper). Note this test was just run at 5V with not a whole lot of effort put into exactly centering/spacing the magnet.

Data can be found here: https://docs.google.com/spreadsheets/d/1EG3RP1kZXmzleNbZcZI58o96MoQLjMxshMYBuX7mUSY/edit?usp=sharing

This data also does not answer the first part of your comment as the encoder is 14 bit (0.088 deg accuracy) and each microstep in this case is only 0.0035 degrees

Thanks,
Josh

Cyrille-de-Brebisson, 2024-10-15:
Woha, thanks!

I will be looking at the data and commenting back here, but thank you so much for doing the work!

Cyrille

joshr120, 2024-10-16:
The difference between your "Series 1" and "Series 2" was just because of my hastily written test code which had a routing error accumulating for each step taken. The "Series 2" in your graph is the correct error measurement. I have fixed the code for these next tests.

The longer oscillation is the most likely due to the sensor not being aligned with the magnet very well. Because this stays consistent for each setup (and returns to 0 after each rotation) it can be easily calibrated out by running the motor with no load and recording the expected position vs encoder reading. 

I have re-run the test with a simple calibration for comparison. The calibration was very simple and just a lookup table of encoder reading vs "actual" position.

The data has been added to a new tab in the [spreadsheet](https://docs.google.com/spreadsheets/d/1EG3RP1kZXmzleNbZcZI58o96MoQLjMxshMYBuX7mUSY/edit?usp=sharing).

And here is the a graph showing error with and without calibration.
[image]

I am not aware of any higher resolution drop in replacements for the AS5600 as most higher resolution encoders use a SPI or alternative interface to also get additional bandwidth.

Cyrille-de-Brebisson, 2024-10-16:
Hello,

1.2° of error seems A LOT!

But who cares, once it is calibrated, it indeed drops to what amounts to 0! Great job...

Thanks for taking the measurements..
Cyrille

closed 2025-05-28 by joshr120