joshr120/PD-Stepper #10: Ethernet question

under-score, 2024-10-30:
great device. Can I attach an ethernet/USB C adapter and use  incoming PTZ commands for motor movements?

joshr120, 2024-10-30:
It may be technically possible but probably not very feasible. The ESP32 would need to act as USB host and you would need to write drivers for the specific USB-Ethernet chip.
You would also need some sort of UDB PD power injector to power the stepper driver at the same time.

closed 2024-10-31 by under-score

joshr120, 2025-12-04:
What does the "PD Status" show on the webserver?

joshr120, 2025-12-04:
Looks like the motor driver is being disabled due to the PD voltage not being negotiated successfully. Are you using a USB C to C cable or an A to C cable?

under-score, 2025-12-04:
A -> C (its an older version)