joshr120/PD-Stepper #15: arduino fails to coompile

amauryverschooren, 2024-11-28:
https://github.com/joshr120/PD-Stepper/blob/614ef368d68fe608b0ca06e0aee6fc43efc8a80e/Software/PD_Stepper_Web_Server/PD_Stepper_Web_Server.ino#L281

https://github.com/joshr120/PD-Stepper/blob/614ef368d68fe608b0ca06e0aee6fc43efc8a80e/Software/PD_Stepper_Web_Server/PD_Stepper_Web_Server.ino#L292

these 2 lines should have const before them.

joshr120, 2024-11-30:
Have you modified the code at all? Seems to compile fine for me in both Arduino IDE 1 and 2

joshr120, 2025-01-09:
I have done some testing and found it likely comes from using the wrong ESPAsyncWebServer library.

I replicated the error when using a fork of the original library: [https://github.com/mathieucarbou/ESPAsyncWebServer](https://github.com/mathieucarbou/ESPAsyncWebServer)
Compiles fine using the original library: [https://github.com/me-no-dev/ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer)

I suggest either change libraries or add const to lines 281 and 292 as @amauryverschooren suggested

Let me know how you get on

amauryverschooren, 2025-01-09:
Hi Josh 

Switching to the other library fixed it for me! 

Thanks for the help!

closed 2025-01-09 by amauryverschooren