joshr120/PD-Stepper #18: [Question] Can I power the module with 12V dc?

schmelly, 2025-01-04:
Dear Josh,

maybe a slightly dumb question but I can I power the module with a 12V dc supply instead of a usb-pd supply?

Background: I intend to use the module to drive a stepper motor for autofocusing an astrophotography setup. Here, typical supply voltages are 12V dc and an approriate connector would be available already in my setup.
If a dc supply won't work this wouldn't be a problem either. I just would have to route an additional usb-c cable to my telescope which would be just a bit more inconvenient.

Thanks and best regards, David

joshr120, 2025-01-04:
Hi David,

There are solder pads on the underside of the PCB where you can directly connect a DC supply of upto 20V. 
[image]

Just make sure you do do not have both the USB and DC power connected at the same time.

Thanks,
Josh

schmelly, 2025-01-05:
Thanks for your prompt support Josh! :-)

Best regards, David

closed 2025-01-05 by schmelly