joshr120/PD-Stepper #19: Please document motor current limits

dwmw2, 2025-01-13:
Thanks for this project; it's great. I've just received three of them.

I'm trying to understand the current settings, so I can hook them up to my (2.4V 1.5A 1.6Ω) stepper motors. I was hoping to find something about that in the QuickStart documentation.

Trying to work it out for myself... you have external 100mΩ resistors on BRA/BRB for Rsense. So according to the table in §8 of the TMC2209 datasheet, that gives a max RMS current of 1.92A, when the VREF input is >=2.5V and IRUN is 31.

[image]

I think the 'current percentage' in the web interface corresponds to IRUN, with 100% mapping to the value 31.

And it looks like the R10 potentiometer allows me to scale VREF from about zero to 2.5V, and you ship it set in the middle, which should mean 100% is 0.96A?
[image]

So for a 1.5A motor I can either set R10 to its maximum and configure the PD Stepper for 78% (25/32), or actually try to precisely set R10 to give me a VREF of 1.95V?

Without putting probes on the board there's no way to actually read what I've set R10/VREF to, is there?

dwmw2, 2025-01-13:
Hm, doesn't clearing GCONF.i_scale_analog mean VREF is ignored... and doesn't the TMC2209 component do exactly that? https://github.com/slimcdk/esphome-custom-components/blob/master/esphome/components/tmc2209/tmc2209_component.cpp#L42

Does that mean VREF from R10 is ignored in the software that's shipped by default?

slimcdk, 2025-01-13:
> clearing GCONF.i_scale_analog mean VREF is ignored...

Correct. VREF is ignored with `I_scale_analog = 0` and current scaling is controlled through `IRUN` and `IHOLD` registers via the [`tmc2209.currents`](https://github.com/slimcdk/esphome-custom-components/tree/master/esphome/components/tmc2209#tmc2209currents-action) action. `I_scale_analog=1` will use VREF and ignore values in said registers. You can overrule that initial setting by [writing to the register](https://github.com/slimcdk/esphome-custom-components/tree/master/esphome/components/tmc2209#advanced) from an `on_boot` action, if you wish to experiment. 

Having the input on VREF for a hardware project like this adds more flexibility for the end user. Anyway, feel free to open a ticket on my project for further discussion on adding in this configuration option.

dwmw2, 2025-01-13:
> `I_scale_analog=1` will use VREF and ignore values in said registers. 

Hm, is that true? I thought `IRUN` and `IHOLD` were still applicable whether `I_scale_analog` is set or not.
[image]

> You can overrule that initial setting by [writing to the register](https://github.com/slimcdk/esphome-custom-components/tree/master/esphome/components/tmc2209#advanced) from an on_boot action, if you wish to experiment.

Your docs tell me that doing that may cause instability in the ESPHome component and should be avoided :)
It also looks like you have a `CONF_ANALOG_SCALE` defined in the config already; I'm just not sure it's wired up.

In fact I don't think I *want* to use VREF, as I have no simple way of measuring what I've set it to. For now I'm just trying to understand how to set the thing up. I *think* that it's configured to ignore VREF by default, so I should just set the current percentage to 70% and that should do the right thing.

dwmw2, 2025-01-13:
Ah, the software on the device as it's shipped isn't ESPHome; it's [this](https://github.com/joshr120/PD-Stepper/tree/main/Software/PD_Stepper_Web_Server) one which uses https://github.com/janelia-arduino/TMC2209. So maybe VREF *is* applicable to the device as it's shipped? So I should probably turn it to max and then set IRUN to 70% (and check it doesn't catch fire).

joshr120, 2025-01-15:
My understanding it that in both the default webserver case as well as when using ESPHome with the component by @slimcdk I_scale_analog is set to 0 and the VREF is ignored. 

I have verified for both cases by adjusting the potentiometer to it's extremes and feeling no difference in the force required to stall the motor.

I will update the getting started guide with a bit more info on how to select the current. I may also update the webserver to give values in the drop down instead of a percentage.

As a side note if you do want to measure VREF there is a test point on the underside of the board.

dwmw2, 2025-01-15:
Thanks! The dropdowns in the webserver probably shouldn't be in multiples of 10% either, should they? Each step in `IRUN` is actually about 0.06A, so the options are:

 - 3% / 60mA
 - 6% / 120mA
 - 9% / 180mA
 - 12% / 240mA
 - 15% / 300mA
 - 18% / 360mA
 - 21% / 420mA
 - 25% / 480mA
 - 28% / 540mA
 - 31% / 600mA
 - 34% / 660mA
 - 37% / 720mA
 - 40% / 780mA
 - 43% / 840mA
 - 46% / 900mA
 - 50% / 960mA
 - 53% / 1.02A
 - 56% / 1.08A
 - 59% / 1.14A
 - 62% / 1.20A
 - 65% / 1.26A
 - 68% / 1.32A
 - 71% / 1.38A
 - 75% / 1.44A
 - 78% / 1.50A
 - 81% / 1.56A
 - 84% / 1.62A
 - 87% / 1.68A
 - 90% / 1.74A
 - 93% / 1.80A
 - 96% / 1.86A
 - 100% / 1.92A

That's not taking a 10% safety margin into account, and it's assuming all current values are in RMS not peak. For my 1.5A motor (which I'm assuming is 1.5A RMS) I'll choose the 71% option, `IRUN=0x16`.

dwmw2, 2025-01-15:
> As a side note if you do want to measure VREF there is a test point on the underside of the board.

Could that be run to an ADC input pin on the ESP32? Or would that cause too much noise?

joshr120, 2025-01-16:
It probably could be fed into an ADC but as it is disabled for most use cases probably not worth it

slimcdk, 2025-02-13:
I spend a bit of time investigating this further (while also adding the option [`analog_current_scale`](https://github.com/slimcdk/esphome-custom-components/blob/master/esphome/components/tmc2209/README.md#stepper-configuration) to set `I_scale_analog` for the ESPHome component) and has now concluded the following. 

* `I_scale_analog=0`  ignores any input from VREF and solely scale currents from what is set in IRUN register. 
* `I_scale_analog=1` appear to take both VREF _and_ IRUN into account such that VREF scales from 0 to what is in the register. So, VREF at 2.5V is equal to what is set in the register. 

Have you tested this and could perhaps confirm my experience? This is not the behavior I understand from reading the docs.

dwmw2, 2025-02-13:
I think that *was* the behaviour I understand from reading the docs, as cited in https://github.com/joshr120/PD-Stepper/issues/19#issuecomment-2587626229 above?

slimcdk, 2025-02-13:
Maybe I read it wrong, but I interpret the docs (and your comment) as "VREF determines the scaling of IRUN" rather than "VREF scales _up to_ IRUN", if that makes any sense. I would argue there is a key distinction here: in the latter case, VREF alone couldn't set full current unless IRUN is at its maximum (31). In contrast, if VREF directly scales the current, it could do so regardless of IRUN's value. That’s why I originally argued that `I_scale_analog` configured either VREF or IRUN scaling determines the current, but what I've later experienced isn't the case. Perhaps you can find time to replicate what I've experienced?

Anyway, I don't think VREF should be used, unless there is a setup where the hardware was specifically designed for it, which isn't the case in this project.

slimcdk, 2025-02-13:
Yep, in either case IRUN will always scale the current. If `I_scale_analog=1`, then VREF adds onto that.

closed 2025-03-27 by joshr120