joshr120/PD-Stepper #20: Help With Stemma

Phando, 2025-01-18:
Is there an example of using a rotary via the stemma connector?
Do I need to set up the stemma pins in my platformio project? If so, what pins are used in the stemma connector.

Thank you!

joshr120, 2025-01-18:
For rotary encoder examples this will very much depend on what module/device you are connecting to it so there are not any examples as part of this project. You will want to refer to the module you are using as to how to use it.

The STEMMA QT connector uses I2C and the pins are as follows:
```
#define SCL     9
#define SDA     8
```

This is already setup in most of the examples as it is on the same bus as the built in AS5600 encoder.

There is not a custom board definition for the PD Stepper but you should just be able to use a generic ESP32-S3 dev board if you would like.

Phando, 2025-01-18:
Thanks for the quick reply! I  using the adafruit led rotary encoder. I’ll check more examples.

closed 2025-02-11 by joshr120