joshr120/PD-Stepper #22: Reverse voltage protection

dwmw2, 2025-02-11:
With the stepper PCB unpowered, if I pull the blind cord manually the motor appears to provide power and the green LED lights up.

With the stepper PCB  powered but idle (in freewheeling mode), the board seems to crash if I pull the blind cord manually above a certain speed.

With the stepper PCB powered from my laptop... my laptop now appears to be completely dead and won't power up at all any more

joshr120, 2025-02-11:
Stepper motors generate back EMF when being back driven and act as a generator. This is very common and devices with stepper motors such as 3D printers should never be back driven too fast. The PD stepper is no exception to this.

Some servo motors will have a braking resistor to soak up some of this generated energy. A large TVS diode on the power rail can also help with shorter spikes but will only be so effective for a continuous reverse power generation.

If the back EMF is over 22V then the buck converter may not longer be able to effectively regulate the 3V3 rail. This is likely what is causing the board to crash in your second scenario once the motor reaches a certain speed.

I will add more warning notes throughout the documentation not to back drive the motor quickly. This is especially the case when a computer is connected, in general a motor should not be spun when powered from a computer (The Arduino sketches all disable the stepper driver unless powered via PD to avoid drawing too much current from a computer). 

If you have blown up a motor board let me know and I can send out a new one.

dwmw2, 2025-02-11:
Is it possible to gate the connections from the motor to the TMC2209 with MOSFETs, so they're only connected when it's actually trying to *drive* the motor?

I think the board is OK, but anyway you've already sent me a spare motor board, thanks! I just ordered a fourth so I have a spare, after noticing that one of my first three has one of the buttons missing (probably from my repeated assembly/disassembly of 3D mount parts).

I'm more worried about the laptop at this point. :)

Arguably *it* should have better protection on its USB port too.

Could the PD stepper board at least have a diode to prevent killing whatever is on the USB port? If the stepper PCB itself dies that's slightly less of an issue.

dwmw2, 2025-02-11:
FWIW I managed to recover the laptop after a few attempts to reset it.

dwmw2, 2025-02-11:
Would setting the tmc2209 `standstill_mode: coil_short_ls` help, at least in the case where the board is powered?

If not, assuming the back EMF is coming through VBUS from the tmc2209 wouldn't it be possible to put a diode there to prevent it feeding back to either the buck converter or the USB connector?

I'd probably be inclined to put *another* diode between the buck converter and the USB connector, for the case where power is being fed to the pads on the underside of the board, if we can stand the voltage drop.