joshr120/PD-Stepper #26: ESPHome integration loses settings after power cycle

inclementweather, 2025-04-15:
Hello,
I'm having a bit of trouble with my PDSteppers.  To set them up initially I flashed them with ESPHome base build from https://web.esphome.io/?dashboard_wizard, then adopted into HA/ESPHome.  After that I copy the advanced blinds config from this repo to the builder interface for the device, compile then flash over Wifi. 

Everything works fine after the flashing process.  I can control the stepper from my HA interface.  However if I remove power and plug it back in it appears to revert back to the initial image I flashed from esphome.io.  

Any idea what might be happening?

joshr120, 2025-04-16:
Hi, when you say it "appears to revert back to the initial image" what exactly is happening which indicates this? 

Additionally, have you tried flashing over USB instead of WiFi?

inclementweather, 2025-04-16:
After the power cycle the ESP appears as online but no longer reports any attributes or settings.  If I check the logs over WiFi it only shows the connection to WiFi being established.  All the normal output from the configured firmware is absent. I can manually turn the stepper and there's no messages about the rotation, no reporting of voltage, etc. 

I only flashed the initial ESPHome firmware over USB but not the configured firmware. I can try doing that later today.

joshr120, 2025-04-16:
Sounds very similar to this problem which was solved by setting a static IP 
https://www.reddit.com/r/Esphome/comments/117x87n/first_esp_device_having_to_reinstall_after_every/

inclementweather, 2025-04-17:
Weird, that was exactly the problem but I'm still not sure why.  Maybe conflicting names stored in home assistant?

closed 2025-04-17 by joshr120