joshr120/PD-Stepper #27: Problem request

mmpzhennan, 2025-04-24:
Hello, I encountered some issues during the process of reproducing this project. After completing the hardware soldering, there was no response when connecting to the computer's USB via type-c. The port of the arduino was gray. Is this a driver problem or a hardware issue?

joshr120, 2025-04-24:
Hi, bit hard to diagnose your problem without more information. 

To start does the 3.3V LED turn on when power is connected?

And what do you mean by "The port of the arduino was gray"?

mmpzhennan, 2025-04-25:
Hello, I have completed this issue. It was caused by my welding process. When I re-welded, the computer could recognize the USB. Thank you very much for your reply

closed 2025-04-25 by joshr120