joshr120/PD-Stepper #28: Serial Commands over USB-C

nirokato, 2025-05-22:
Hello,

Thank you for creating such a great platform. It's so cool!

On to the issue:
> Serial commands can be sent/received through the USB-C connector (Note if your device sending the commands cannot provide enough power you may need an injector of some kind)

I went through as much of the documentation as possible, and I couldn't really find anything about actually sending commands over usb-c.

Setup:
I'm successfully powering the pd-stepper using a Pixel 8 Pro's OTG (pd-stepper set to 5V, 30% max current) and using [Serial USB Terminal](https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal&pcampaignid=web_share) to view logging output: 

Here's the pd-stepper web settings:
[image]

Here's the Serial USB Terminal interaction, keep in mind the slider was set to Snap Centre:
```
21:36:59.286 Connected to CDC device
21:36:59.292 Code Starting
21:36:59.292 AP IP address: 192.168.4.1
21:38:43.202 
21:38:52.612 Slider value: 0
21:38:53.431 Slider value: 207
21:38:53.513 Slider value: 0
21:38:58.570 Slider value: 207
21:38:58.657 Slider value: 0
21:38:59.289 Slider value: 207
21:38:59.377 Slider value: 0
21:39:14.465 help
21:48:24.384 h
21:48:26.615 ?
21:48:32.983 status
21:48:36.996 info
21:48:41.961 Disconnected
21:48:43.129 Connected to CDC device
21:48:43.129 Build:Mar 27 2021
21:48:58.693 a
21:49:00.038 ab
21:49:02.008 b
```
As you can see, I blindly sent commands and didn't receive any output from the pd-stepper. 

What is a hardware newb like me to do next?

Thank you!

joshr120, 2025-05-22:
Hi,
Cool idea to connect it directly to a phone! Currently none of the examples have been setup to receive serial commands. You would need to modify one of the examples to respond to your commands.

However, I am currently working on a USB hub with USB PD on the downstream (device) ports which will allow serial control. As part of this I will be adding a serial control example.

Matoseb, 2025-07-05:
Hi, I'm not sure why I can't receive any Serial messages when I'm simply connected to my macbook via USB-C. I'm running a basic Serial sketch, with small delays before Serial.begin. But I can't receive any messages.
Am I missing something?

Matoseb, 2025-07-05:
> Hi, I'm not sure why I can't receive any Serial messages when I'm simply connected to my macbook via USB-C. I'm running a basic Serial sketch, with small delays before Serial.begin. But I can't receive any messages. Am I missing something?

Got it! In the board settings in Arduino, USB CDC On Boot must be set to "Enabled"

Matoseb, 2025-08-01:
> However, I am currently working on a USB hub with USB PD on the downstream (device) ports which will allow serial control. As part of this I will be adding a serial control example.

Would this usb hub allow to safely flash the esp32 while it has external power connected to its vbus solder pad?

joshr120, 2025-08-02:
Even with the custom hub, the board should never be externally powered by the VBUS solder pads and connected to USB at the same time. 
If you absolutely needed to, you could cut open a USB cable and disconnect the red VBUS wire inside the cable.