joshr120/PD-Stepper #30: No access point is created

emgiorzan, 2025-05-23:
Hi all,

After successfully uploading the code provided @https://github.com/joshr120/PD-Stepper/tree/main/Software/PD_Stepper_Web_Server

I reset the module and LED4 stays green while LED3 is blinking.

On my laptop I cannot see any access point..

[image]

Any suggestions?

joshr120, 2025-05-24:
Hi, if LED 3 is blinking this seems like a USB PD power/negotiation issue. 

Some more info would probably help. How are you powering the board? do you have a motor plugged in etc.

Before uploading the code yourself did you manage to connect to the webserver with the pre-installed code?

emgiorzan, 2025-05-24:
> If LED4 is blinking that is a power issue. Does it still happen when there
> is no motor connected?

Unfortunately I am not able to test it today. Possible back on it on Sunday
or Monday. Is there a way to reset it?

joshr120, 2025-05-24:
Does not necessarily sounds like a software issue. You can  however force it into bootloader mode by holding down the BOOT button, pressing RST and then releasing BOOT. You can then re-flash the firmware. The firmware you have loaded should be exactly the same as what was shipped on it.

joshr120, 2025-05-25:
Very hard to know without more information. I would first try with disconnecting the motor and a different power supply.

emgiorzan, 2025-05-26:
Thanks,
After checking the board alone, it will compile and run, but for the motor,
I have a faulty adaptor and not enough voltage to operate the motor.